adafruit/MAX6675-library #5: Errors in code!?

yosef321, 2015-06-04:
when i run the code and library i get this error " fatal error: util/delay.h: No such file or directory
 #include <util/delay.h>
                        ^
compilation terminated.
Error compiling.
". Not sure if i missed something or something is wrong with the code.

pit001, 2015-09-08:
dito, also the link.

// this example is public domain. enjoy!
// www.ladyada.net/learn/sensors/thermocouple

is 404

pit001, 2015-09-08:
the file library.properties is empty, 
fill in this:
name=Adafruit MAX6675
version=1.0.0
author=Adafruit
maintainer=Adafruit <info@adafruit.com>
sentence=MAX6675
paragraph=MAX6675
category=Uncategorized
url=https://github.com/adafruit
architectures=*

florinbuda85, 2019-10-03:
4 years later i have the same issue..

sparth62, 2019-10-04:
> 4 years later i have the same issue..

Hi friend, I successfully solved this issue, Open max6675.cpp file, Remove #include <util/delay.h> and replace all _delay_ms(1); with delay(1);..... save and run code

ladyada, 2020-06-04:
fixed

closed 2020-06-04 by ladyada