adafruit/MAX6675-library #9: Problem with ESP8266

ppotok, 2015-12-30:
Hello,
When I want to compile library with NodeMCU V1 board I got error: 
MAX6675-library-master/max6675.cpp:9:24: fatal error: util/delay.h: No such file or directory
 #include <util/delay.h>
                        ^
compilation terminated.

Please is there any possibility to fix it?
Thanks!

donatoaz, 2015-12-31:
I had the same problem and I managed to get it to compile by changing max6675.cpp to:

    ...
    #elif defined(ESP8266)
      #include <pgmspace.h>
      #define _delay_ms(ms) delayMicroseconds((ms) * 1000)
    #endif
    
    //#include <util/delay.h>
    #ifdef __avr__
      #include <util/delay.h>
    #endif
    ...

bauer-san, 2016-01-01:
ppotok,
Have a look at my open pull request: https://github.com/adafruit/MAX6675-library/pull/8

I think it solves the issue too.

SirUli, 2016-01-26:
@donatoaz: Can you share how you connected the ESP-01 to the MAX6675?

I tried:

MAX6675| ESP8266
-------|------------------
CS | GPIO2
SCK | GPIO0
SO | GPIO3 (RX)

And configured the code that way (and included your fix above):
```
int thermoDO = 3;
int thermoCS = 2;
int thermoCLK = 0;

MAX6675 thermocouple(thermoCLK, thermoCS, thermoDO);
```

Thanks in advance!

Best Regards,
Uli

donatoaz, 2016-01-29:
@SirUli I published my code on github. 

[esp8266-esp01-MAX6675](https://github.com/donatoaz/esp8266-esp01-MAX6675)

I take it you have tested your temperature probe and it is working.

I remember having some issues with using the RX pin as an input, something to do with bootup modes on the esp01... I am trying to find the table that shows the modes but cant find it...

SirUli, 2016-01-29:
@donatoaz i forgot to set the pin-mode... oh man. Well now it works although i had to change the pinmode to: pinMode(3, INPUT);

Not sure if that works for you, but that was at least valid in my case :) Thanks again for your quick help!!

donatoaz, 2016-01-29:
@SirUli good to know.

Mine was already working. Nut I decided to switch to DS18s20, simply for cost reasons...

I need to learn how to use github so I can submit a pull request to the owner of this repo.

SirUli, 2016-01-29:
@donatoaz well i haven't found a different sensor which is applicable to my gas-grill - so will have to stick with the MAX6675 ;)

SirUli, 2016-06-23:
I [forked the library](https://github.com/SirUli/MAX6675) as Adafruit doesn't update anymore. Just if anyone else needs a ESP8266 compatible library.

tioraulv, 2016-09-29:
Using 
Serial.begin(115200,SERIAL_8N1,SERIAL_TX_ONLY);
allows you to use RX as a general purpose input (or another output), while still writing debug messages to Serial. Again the 330 ohm resistor in the RX lead to the Flash programmer protects against shorting out the programmer's driver.
Source http://www.forward.com.au/pfod/ESP8266/GPIOpins/ESP8266_01_pin_magic.html

stephenmayer, 2017-01-15:
donatoaz thank you!  Your suggestion worked great for me too! Appreciate you sharing ...

fixduino, 2018-01-31:
thanks for this isue. i will try max 6675 with nodemcu,, and hav same trouble with "pgmspace and util/delay".

have edit max6675.h with

`#if defined(ESP8266)
  #include <pgmspace.h>
  #define _delay_ms(ms) delayMicroseconds((ms) * 1000)
#endif

//#include <util/delay.h>
#ifdef __avr__
  #include <util/delay.h>
#endif

#include <stdlib.h>
#include "max6675.h"`

and get success compiled, .

GhostBug-007, 2018-04-15:
Please fix the issue and merge this pull request.

zobotek, 2020-01-20:
IMHO, a better solution is to write an esp8266-specific util/delay.h and add a pull request to include it in the esp8266 board distribution.

s00500, 2020-05-22:
The same fix needs to be applied for the ESP32 btw...

ladyada, 2020-06-04:
fixed

closed 2020-06-04 by ladyada