adafruit/MAX6675-library #22: Error when the sensor is disconnected.

Samodelkin-YouTube, 2021-01-02:
Hi, you can add a variable to the library that will change its status? For example, if the connection is lost or the thermocouple is damaged during operation.
Thanks.

ladyada, 2021-01-02:
the max6675 does not have good error handling...if you want to add something we'll look at a PR for sure! however, this chip is disconitnued and people use the max31855 or others

zaphodikus, 2021-12-05:
Plenty of 7765 chips available, and for most people's use the 6675 accuracy is perfectly good no need to spend more on a 31855 for accuracy in a setup where it's not needed.

Basically if you get the LSB masked with 0x04 == 0x04, then it's an error. That's the status bits in the word that are unused for data. Or if you read some insane value like 0 degrees (quite possible) or a value >500 degrees, that's an error.  I would put in a PR, but this is a bit-banged library so pretty low interest value to start with might want to find a library that uses the clock and register on-board the chip